SICKAG/sick_safetyscanners2 #1: Cannot install from source

SomaGallai, 2021-01-12:
Ubuntu 20.04.1 LTS
Nvidia Jetson  TX2

Following the steps given, the build fails.
The error it gives is the following:
```
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findsick_safetyscanners_base.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "sick_safetyscanners_base", but CMake did not find one.

  Could not find a package configuration file provided by
  "sick_safetyscanners_base" with any of the following names:

    sick_safetyscanners_baseConfig.cmake
    sick_safetyscanners_base-config.cmake

  Add the installation prefix of "sick_safetyscanners_base" to
  CMAKE_PREFIX_PATH or set "sick_safetyscanners_base_DIR" to a directory
  containing one of the above files.  If "sick_safetyscanners_base" provides
  a separate development package or SDK, be sure it has been installed.
```

puck-fzi, 2021-01-12:
Hi,

did you include https://github.com/SICKAG/sick_safetyscanners_base in your colcon_ws? This is the missing dependency. 
The other dependency is https://github.com/SICKAG/sick_safetyscanners2_interfaces. Currently both have to be in your workspace. A release is triggered as apt package

SomaGallai, 2021-01-12:
Thank you for the clarification, now it was able to build!

closed 2021-01-12 by SomaGallai

puck-fzi, 2021-01-12:
Great, I'll clarify it in the build instruction.