SICKAG/sick_safetyscanners2 #4: order of fields in ~/output_paths

eliasdc, 2021-02-24:
We want to adjust our mobile's robot based on the active fields in a specific monitoring case. I've been looking into the topic `output_paths` but the order looks weird. All the field sets have WF1, WF2, PF1, PF2 fields configured (in that order) with W standing for warning and P standing for protective. The numbers stand for which field would be reached first.

The 4 first array values in all properties of output_paths correspond to these fields but in a different order. Based on the `is_safe` property I can clearly see that index 1 and 3 is PF1 and PF2 or in reverse order (I can't know) and index 0 and 2 is WF1 and WF2. My problem is that I can't easily see how this order is defined.

In the sick safety designer software I can see the order for the current active monitoring case when I generate the report. The section `7.6 Fields` defines the order for monitoring case 1 as WF2, PF1, WF1, PF2 (which I don't know where it's coming from as it is different from the order in the field sets section 6.7.1). As the report doesn't mention the fields order for none active monitoring cases I'm unable to define the order for all monitoring cases.
[Report_2021-02-24_13-29-42.pdf](https://github.com/SICKAG/sick_safetyscanners2/files/6035934/Report_2021-02-24_13-29-42.pdf)

puck-fzi, 2021-03-05:
Hi, sorry for the late reply.

I will have a look into this and see if I can find any specific order. The order of the fields should be in the order the scanner sends them to the driver. But I will need to have a look into how this order is generated from driver side.

puck-fzi, 2021-03-19:
Hi Elias, 
I was in contact with SICK regarding your Problem.
It seems that his is a device-related issue, not driver. Please contact the product manager from SICK: maximilian.nickola[at]sick.de

closed 2021-03-19 by puck-fzi

eliasdc, 2021-03-31:
I figured out a way to identify the order or change it. By exporting the monitoring case table to xml and looking at the `<Q>` tag (starting from 1). Altering the order to what you want and importing it again fixes the order. It is weird that it does not correspond to the display order in safety designer or that you can not specify it in the UI.

puck-fzi, 2021-03-31:
Thanks for providing your solution.