SICKAG/sick_safetyscanners2 #6: Unable to use use_persistend_config:=false

eliasdc, 2021-03-26:
When using a NanoScan3 we're unable get messages from the `/scan` topic when trying to configure `angle_start` and `angle_end`. These settings are only pushed to the lidar if `use_persistend_config:=false` but from that moment we don't receive messages on the topic. With `use_persistend_config:=true` everything works except for the changed angles of course. Changing this angle in the sick safety designer does not work. In the data output option the angles selection boxes are greyed out.

It does work with our custom `sick_safetyscanners` port (https://github.com/Tractonomy/sick_safetyscanners) to ros2 with the same settings, just altering the node and package in the launch file.

```
[INFO] [sick_safetyscanners2_node-1]: process started with pid [8642]
[sick_safetyscanners2_node-1] [INFO] [1616761017.855698792] [sick_safetyscanners2_node]: Initializing SickSafetyscannersRos2 Node
[sick_safetyscanners2_node-1] [INFO] [1616761017.855908010] [sick_safetyscanners2_node]: frame_id: top_lidar_sensor_frame
[sick_safetyscanners2_node-1] [INFO] [1616761017.855920080] [sick_safetyscanners2_node]: sensor_ip: 172.29.29.11
[sick_safetyscanners2_node-1] [INFO] [1616761017.855945050] [sick_safetyscanners2_node]: host_ip: 172.29.29.1
[sick_safetyscanners2_node-1] [INFO] [1616761017.855951561] [sick_safetyscanners2_node]: host_udp_port: 0
[sick_safetyscanners2_node-1] [INFO] [1616761017.855957664] [sick_safetyscanners2_node]: channel: 0
[sick_safetyscanners2_node-1] [INFO] [1616761017.855963329] [sick_safetyscanners2_node]: channel_enabled: true
[sick_safetyscanners2_node-1] [INFO] [1616761017.855981586] [sick_safetyscanners2_node]: skip: 0
[sick_safetyscanners2_node-1] [INFO] [1616761017.855987224] [sick_safetyscanners2_node]: angle_start: -2.100000
[sick_safetyscanners2_node-1] [INFO] [1616761017.856013327] [sick_safetyscanners2_node]: angle_end: 2.100000
[sick_safetyscanners2_node-1] [INFO] [1616761017.856019637] [sick_safetyscanners2_node]: time_offset: 0.000000
[sick_safetyscanners2_node-1] [INFO] [1616761017.856026554] [sick_safetyscanners2_node]: general_system_state: true
[sick_safetyscanners2_node-1] [INFO] [1616761017.856032865] [sick_safetyscanners2_node]: derived_settings: true
[sick_safetyscanners2_node-1] [INFO] [1616761017.856038956] [sick_safetyscanners2_node]: measurement_data: true
[sick_safetyscanners2_node-1] [INFO] [1616761017.856044913] [sick_safetyscanners2_node]: intrusion_data: true
[sick_safetyscanners2_node-1] [INFO] [1616761017.856051107] [sick_safetyscanners2_node]: application_io_data: true
[sick_safetyscanners2_node-1] [INFO] [1616761017.856079497] [sick_safetyscanners2_node]: use_persistent_config: false
[sick_safetyscanners2_node-1] [INFO] [1616761017.856087736] [sick_safetyscanners2_node]: min_intensities: 0.000000
[sick_safetyscanners2_node-1] [INFO]: Command Method Acknowledged.
[sick_safetyscanners2_node-1] [INFO] [1616761017.859604478] [sick_safetyscanners2_node]: Communication to Sensor set up
[sick_safetyscanners2_node-1] [INFO] [1616761017.859660753] [sick_safetyscanners2_node]: Reading Type code settings
[sick_safetyscanners2_node-1] [INFO]: Command Variable Acknowledged.
[sick_safetyscanners2_node-1] [INFO]: Type Code: NANS3-CAAZ30AN1P04
[sick_safetyscanners2_node-1] [INFO] [1616761017.862106282] [sick_safetyscanners2_node]: Node Configured and running
```

puck-fzi, 2021-03-30:
Hi, I will try to debug this. Thanks for bringing this to my attention

puck-fzi, 2021-03-30:
Alright i can reproduce the issue. However with a microScan3 it is behaving as expected. Just the nanoScan3 does not behave like it is supposed to.
Furthermore, I fixed a small typo in the default launch file.

eliasdc, 2021-03-31:
I completely missed that typo, fixed in initial post too for future searches. I was however using the correct parameter when testing from my own launch file.

Just tested your fix and it does work as expected. Thanks.