SICKAG/sick_safetyscanners2 #8: Multiple package launch makes Sick_safetyscanners2_node-1 crash

SomaGallai, 2021-05-18:
Hi,

We have been struggling with an issue regarding Sick_safetyscanners2 node. Standalone only launching Sick_safetyscanners2 30 times has no issues, but using our launch file running multiple nodes after 16 times, Sick_safetyscanners2 gives process has died error with exit code -11. The weird part is that the error shows up after the Communication to Sensor set up has already occured

`
[sick_safetyscanners2_node-1] [INFO]: Command Method Acknowledged.
[sick_safetyscanners2_node-1] [INFO] [1621334701.382357271] [testing.sick_safetyscanners2_node]: Communication to Sensor set up
[sick_safetyscanners2_node-1] [INFO] [1621334701.382614487] [testing.sick_safetyscanners2_node]: Reading Type code settings
[sick_safetyscanners2_node-1] [INFO]: Command Variable Acknowledged.
[sick_safetyscanners2_node-1] [INFO]: Type Code: NANS3-CAAZ30AN1P04
`
Only missing INFO is the following when this error occurs:
`[sick_safetyscanners2_node-1] [INFO] [1621336333.044491863] [serena.sick_safetyscanners2_node]: Node Configured and running`

The full error code:
`[ERROR] [sick_safetyscanners2_node-1]: process has died [pid 2355821, exit code -11, cmd '/home/somagallai/user/testing/install/sick_safetyscanners2/lib/sick_safetyscanners2/sick_safetyscanners2_node --ros-args -r __node:=sick_safetyscanners2_node -r __ns:=/testing --params-file /tmp/launch_params_3o1wk_jy'].`

Is there a way to see a more explicit error description what could be causing the issue?

puck-fzi, 2021-05-18:
Hi, 

this is odd behaviour. To Further inspect this build the package with Debug Information from source. So either use `RelWithDebInfo` or `Debug`

```
colcon build <other_args> --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
```

Afterwards you are able to start the node with gdb. This is nicely described in:
https://navigation.ros.org/tutorials/docs/get_backtrace.html

This allows you to get a stack trace and see where the error is occurring.

SomaGallai, 2021-05-20:
Hi,

I have tested our launch file using gdb, thank you for the instructions! Using gdb I was not able to make Sick_safetyscanners2 crash 40+ tries. Without gdb, but using ros2 launch -debug argument I was able to catch that when Sick_safetyscanners2 crashes the order of methods called are messed up. 

When Sick_safetyscanners2 works as expected the method calls are the following:
- Process start
- Initializing
- Parameters configuration
- Command method Acknowledged
- Communication to Sensor Set Up
- Reading Type Code settings
- Command variable Acknowledged
- Type code
- Command Method Acknowledged
- Node configured and running

When it crashes the following method call order has been seen:

crash1:
- Process start
- Initializing
- Parameters configuration
- Command method Acknowledged
- Command variable Acknowledged
- Type code
- Communication to Sensor Set Up
- Reading Type Code settings
- Process has died

crash2:
- Process start
- Initializing
- Parameters configuration
- Communication to Sensor Set Up
- Reading Type Code settings
- Command method Acknowledged
- Command variable Acknowledged
- Type code
- Process has died

puck-fzi, 2021-05-21:
Hm okay, that looks like some sort of racing condition might be happening. IS there anything special in your launch file or your setup? And do you have high latencies between sensor and driver? I will try to have a look into it, but since I haven't experienced the problem myself I might not be able to recreate it. So any insights would be helpful

puck-fzi, 2021-06-30:
Are there any new insights? Did you find a solution for the problem?

SomaGallai, 2021-07-02:
> Are there any new insights? Did you find a solution for the problem?

It seems as PR #11 seems to do the trick, we have been using it for a while and the issue has not arose since then.

puck-fzi, 2021-09-09:
Closing this as PR#11 was merged

closed 2021-09-09 by puck-fzi