SICKAG/sick_safetyscanners2 #9: Lifecycle Node

guilyx, 2021-05-20:
Hi there,

Is there any plans to use lifecycle nodes for the scanner driver? As it may crash, being able to add this to a lifecycle manager along with controllers and other low level critical packages would prevent any robot from moving if one of the nodes die. 
Not sure if you're remotely interested in doing this, but if you were I'd be happy to develop this on a parallel branch maybe ?

puck-fzi, 2021-05-21:
Hi, currently this is not on my agenda.
But please feel free to open a PR for the lifecycle nodes, I am happy to review it and merge it into the driver it it looks good.
In the best case scenario both the lifecycle node and the current not a both on the main branch and can both be used depending on what the user needs.

guilyx, 2021-05-21:
Hi, yeah fine. Someone will shoot a PR next week for this.

puck-fzi, 2021-06-30:
Hi, just a follow up, are you still working on this? if not i would mark it as help wanted

guilyx, 2021-06-30:
Hey, I’m sending the PR today!

puck-fzi, 2021-09-09:
Corresponding PR #11 has been merged into the master

closed 2021-09-09 by puck-fzi