SICKAG/sick_safetyscanners2 #12: ROS2 Galactic binary support

eliasdc, 2021-09-14:
Is it possible to release a binary for the latest ROS2 version Galactic? 

including both dependencies
- sick_safetyscanners_base (https://github.com/SICKAG/sick_safetyscanners_base)
- sick_safetyscanners2_interfaces (https://github.com/SICKAG/sick_safetyscanners2_interfaces)

puck-fzi, 2021-09-15:
Hi Elias,

I'll trigger the release at the beginning of october. Until then i am Not in Office.

Leonngm, 2021-11-12:
Hi @eliasdc,
which environment are you using for ros galactic and did you change anything in the launch files for sick_safetyscanner except of the ip addresses?

Regards

eliasdc, 2021-11-15:
My launch file is the following with custom ips, frame id name and angles: 

```python
#!/usr/bin/env python

import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration

def generate_launch_description():

    frame_id = LaunchConfiguration("frame_id", default="top_lidar_sensor_frame")
    sensor_ip = LaunchConfiguration("sensor_ip", default="172.29.29.11")
    host_ip = LaunchConfiguration("host_ip", default="172.29.29.1")
    host_udp_port = LaunchConfiguration("host_udp_port", default="0")
    channel = LaunchConfiguration("channel", default="0")
    channel_enabled = LaunchConfiguration("channel_enabled", default="true")
    skip = LaunchConfiguration("skip", default="0")
    angle_start = LaunchConfiguration("angle_start", default="-2.1")
    angle_end = LaunchConfiguration("angle_end", default="2.1")
    time_offset = LaunchConfiguration("time_offset", default="0.0")
    general_system_state = LaunchConfiguration("general_system_state", default="true")
    derived_settings = LaunchConfiguration("derived_settings", default="true")
    measurement_data = LaunchConfiguration("measurement_data", default="true")
    intrusion_data = LaunchConfiguration("intrusion_data", default="true")
    application_io_data = LaunchConfiguration("application_io_data", default="true")
    use_persistent_config = LaunchConfiguration("use_persistent_config", default="false")
    min_intensities = LaunchConfiguration("min_intensities", default="0.0")

    return LaunchDescription(
        [
            Node(
                package="sick_safetyscanners2",
                executable="sick_safetyscanners2_node",
                name="sick_safetyscanners2_node",
                output="screen",
                emulate_tty=True,
                remappings=[
                    ('/diagnostics', 'diagnostics')
                ],
                parameters=[
                    {
                        "sensor_ip": sensor_ip,
                        "host_ip": host_ip,
                        "host_udp_port": host_udp_port,
                        "channel": channel,
                        "channel_enabled": channel_enabled,
                        "frame_id": frame_id,
                        "skip": skip,
                        "angle_start": angle_start,
                        "angle_end": angle_end,
                        "time_offset": time_offset,
                        "general_system_state": general_system_state,
                        "derived_settings": derived_settings,
                        "measurement_data": measurement_data,
                        "intrusion_data": intrusion_data,
                        "application_io_data": application_io_data,
                        "use_persistent_config": use_persistent_config,
                        "min_intensities": min_intensities,
                    }
                ],
            ),
        ]
    )
```

Leonngm, 2021-12-03:
Thanks for sharing your launch file. 

May i ask you which system and os you are using, your gcc version and your boost version?
Thanks,
Leon

eliasdc, 2021-12-06:
I'm using Ubuntu 20.04 with gcc version `gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0` and boost `1.71.0.0ubuntu2`.

Leonngm, 2021-12-06:
Okay, we are using the exact same system setup, but for me the drivers won't work. :(

puck-fzi, 2021-12-22:
I got around to triggering the release, with the next sync all three packages should be released for galactic.

closed 2021-12-22 by puck-fzi

NikolasE, 2022-01-03:
I just installed galactic on a new system and the sick-safetyscanners2 was not found (base and interface however were available). 
This fits also to this post: [Discource-post](https://discourse.ros.org/t/new-packages-for-galactic-geochelone-2021-12-27/23607) that only lists these two packages. 

Was the main package really released with the two support packages?

puck-fzi, 2022-01-03:
Hi, yes, the main repo was held for the sync since the dependencies had to be merged first. So the main repo will be included in the next sync. More information here: https://github.com/ros/rosdistro/pull/31566