SICKAG/sick_safetyscanners2 #18: Race condition during devices configuration

MarcoMatteoBassa, 2022-05-16:
Hi guys, thank you for the great work on the ros2 interface. Unfortunately here at Idealworks we came across what we think is a race condition during the configuration of the devices using the lifecycle-managed version of the driver. In particular, if the the calls to configure and activate are performed very quickly one after the other (but of course waiting for the result of the service), the devices don't activate the udp transmission of the data even if their configuration is apparently successfull. The problem is in our case solved if we leave a bit of time in-between the calls (ex 1 second) or if we remove the call to "changeSensorSettings" in the constructor of "SickSafetyscannersBase". The hypothesis here is that two consecutive calls of "changeSensorSettings" can bring the device into an undesired state.

Here is an example of ros-log where the device is activated quickly by a manager and hence doesn't publish UDP data (~0.25 seconds between config and activate)

```
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.550510809] [robot_interface.micro_scan_rear]: Initializing SickSafetyscannersLifeCycle 
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.551965666] [robot_interface.micro_scan_rear]: frame_id: scan_rear
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552033700] [robot_interface.micro_scan_rear]: sensor_ip: 192.168.1.4
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552079461] [robot_interface.micro_scan_rear]: interface_ip: 0.0.0.0
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552113542] [robot_interface.micro_scan_rear]: host_ip: 192.168.1.10
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552146439] [robot_interface.micro_scan_rear]: host_udp_port: 6062
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552181000] [robot_interface.micro_scan_rear]: channel: 0
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552213545] [robot_interface.micro_scan_rear]: channel_enabled: true
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552246378] [robot_interface.micro_scan_rear]: skip: 0
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552278091] [robot_interface.micro_scan_rear]: angle_start: -0.523599
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552325036] [robot_interface.micro_scan_rear]: angle_end: 3.752460
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.552361357] [robot_interface.micro_scan_rear]: time_offset: 0.000000
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.555316032] [robot_interface.micro_scan_rear]: general_system_state: true
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.555419971] [robot_interface.micro_scan_rear]: derived_settings: true
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.555507365] [robot_interface.micro_scan_rear]: measurement_data: true
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.556242714] [robot_interface.micro_scan_rear]: intrusion_data: true
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.556307516] [robot_interface.micro_scan_rear]: application_io_data: true
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.556347133] [robot_interface.micro_scan_rear]: use_persistent_config: false
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.556381310] [robot_interface.micro_scan_rear]: min_intensities: 0.000000
[lifecycle_manager-14] [INFO] [1652691167.556707463] [robot_interface.lifecycle_manager]: Autostarting
[lifecycle_manager-14] [INFO] [1652691167.556804202] [robot_interface.lifecycle_manager]: Starting all nodes
[lifecycle_manager-14] [INFO] [1652691167.556884684] [robot_interface.lifecycle_manager]: Configuring micro_scan_rear
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.624727358] [robot_interface.micro_scan_rear]: on_configure()...
[sick_safetyscanners2_lifecycle_node-13] [INFO]: Command Method Acknowledged.
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.745830439] [robot_interface.micro_scan_rear]: Communication to Sensor set up
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.745949515] [robot_interface.micro_scan_rear]: Reading Type code settings
[sick_safetyscanners2_lifecycle_node-13] [INFO]: Command Variable Acknowledged.
[sick_safetyscanners2_lifecycle_node-13] [INFO]: Type Code: MICS3-CBAZ55ZA1
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.872954042] [robot_interface.micro_scan_rear]: Node Configured
[lifecycle_manager-14] [INFO] [1652691167.874495205] [robot_interface.lifecycle_manager]: Activating micro_scan_rear
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.875722120] [robot_interface.micro_scan_rear]: on_activate()...
[sick_safetyscanners2_lifecycle_node-13] [INFO]: Command Method Acknowledged.
[sick_safetyscanners2_lifecycle_node-13] [INFO] [1652691167.992948485] [robot_interface.micro_scan_rear]: Node activated, device is running...
[lifecycle_manager-14] [INFO] [1652691167.994265546] [robot_interface.lifecycle_manager]: All nodes are active
```

And here is a log where the device is activated properly (for example using 2 consecutive cli commands like:
ros2 service call /robot_interface/micro_scan_front/change_state lifecycle_msgs/srv/ChangeState "{transition: {id: 1}}" && ros2 service call /robot_interface/micro_scan_front/change_state lifecycle_msgs/srv/ChangeState "{transition: {id: 3}}"
ros2 service call /robot_interface/micro_scan_front/change_state lifecycle_msgs/srv/ChangeState "{transition: {id: 3}}")

```
[sick_safetyscanners2_lifecycle_node-12] [INFO] [1652689148.136076956] [robot_interface.micro_scan_front]: on_configure()...
[sick_safetyscanners2_lifecycle_node-12] [INFO]: Command Method Acknowledged.
[sick_safetyscanners2_lifecycle_node-12] [INFO] [1652689148.275378069] [robot_interface.micro_scan_front]: Communication to Sensor set up
[sick_safetyscanners2_lifecycle_node-12] [INFO] [1652689148.276690650] [robot_interface.micro_scan_front]: Reading Type code settings
[sick_safetyscanners2_lifecycle_node-12] [INFO]: Command Variable Acknowledged.
[sick_safetyscanners2_lifecycle_node-12] [INFO]: Type Code: MICS3-CBAZ55ZA1
[sick_safetyscanners2_lifecycle_node-12] [INFO] [1652689148.410164459] [robot_interface.micro_scan_front]: Node Configured
[sick_safetyscanners2_lifecycle_node-12] [INFO] [1652689149.850215740] [robot_interface.micro_scan_front]: on_activate()...
[sick_safetyscanners2_lifecycle_node-12] [WARN] [1652689149.851392798] [LifecyclePublisher]: Trying to publish message on the topic '/robot_interface/scan', but the publisher is not activated
```

In the latter case we also get a bunch of annoying warnings ("Trying to publish message...") before the publisher actually gets activated and we start getting data on the topics :)

I'll also attach a tcp dump of a failing initialization.

[tcp_dump_failure.txt](https://github.com/SICKAG/sick_safetyscanners2/files/8698538/tcp_dump_failure.txt)

Could you guys take a look and confirm that there's something wrong? Thanks :)

puck-fzi, 2022-05-30:
Hi, since i did not directly implement the lifecycle nodes, I am not too sure what exactly is causing your issue, maybe @guilyx has some more insights or maybe experienced the issue himself so far.

What do you mean by stating its working properly using a different driver?

guilyx, 2022-05-31:
Hey there,

We haven't had issues ever since we closed the last PR, we're using lifecycle nodes, I haven't experienced this (and the scanner node is launched and killed hundred times a day here)

closed 2022-06-08 by MarcoMatteoBassa