SICKAG/sick_safetyscanners2 #23: process exits abnormally with range check exception

cmraaron, 2023-03-06:
after four hours of running happily, the sick_safetyscanners2_node process exited with `vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)`

probably not easy to reproduce, its the first time I've seen it.  But worth logging.
Interesting the size of the vector is 0, and the index is probably -1.  Perhaps that offers some clues.
The only code i've seen that looks like that is here
https://github.com/SICKAG/sick_safetyscanners2/blob/68241383e681a615e7d350f0f4066d1368ce1eae/src/utils/MessageCreator.cpp#L179
But it would never have entered the loop if the size was 0

puck-fzi, 2023-04-04:
Thanks for noticing, this seems to be a bug which transferred form the ROS1 version to the ROS2 version.
And yes you are correct, so far I was not able to reproduce the error. I will continue looking and trying to catch the error, however this already proved difficult in the ROS1 version since it happens rarely and so far never on my own system.

reinzor, 2023-10-24:
Still in latest version:

```
[robot_73981][INFO] 2023-10-19 16:15:28,219:   what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

TanmayDeshmukh, 2024-05-06:
Hi, 
We had this exact issue appear once after several hours of running as well.
Could it be coming from deeper within? sick_safetyscanners_base/cola maybe?

TanmayDeshmukh, 2024-05-13:
The ROS2 node itself doesn't crash, and continues to publish diagnostic_msgs, but stops publishing scan data, which suggests that one of the threads spawned internally crashes.
I suspect [this thread](https://github.com/SICKAG/sick_safetyscanners_base/blob/ros2/src/SickSafetyscanners.cpp#L246C1-L257C1) in sick_safetyscanners_base simply disappears after the range check exception. The `try..catch` would explain why it exits "normally".
I'm not familiar with the code-base so just speculating.

It seems to happen usually after over 24 hours. I will setup gdb and try to get a backtrace.

reinzor, 2024-05-13:
@yannickdehoop fyi

TanmayDeshmukh, 2024-05-23:
Here's the stacktrace
Looks like the culprint is this line:
https://github.com/SICKAG/sick_safetyscanners2/blob/4606e96e827ea6726e9936b8f5017813cb53309d/src/utils/MessageCreator.cpp#L62

```
#0  0x00007ffff78649fc in pthread_kill () from target:/lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff7810476 in raise () from target:/lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff77f67f3 in abort () from target:/lib/x86_64-linux-gnu/libc.so.6
#3  0x00007ffff7abbb9e in ?? () from target:/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7ac720c in ?? () from target:/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7ac7277 in std::terminate() () from target:/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7ac74d8 in __cxa_throw () from target:/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7abe4cd in ?? () from target:/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00005555555b09bc in std::vector<sick::datastructure::ScanPoint, std::allocator<sick::datastructure::ScanPoint> >::_M_range_check (this=0x7fffe67fa5e0, this=0x7fffe67fa5e0, __n=<optimized out>)
    at /usr/include/c++/11/bits/stl_vector.h:1073
#9  std::vector<sick::datastructure::ScanPoint, std::allocator<sick::datastructure::ScanPoint> >::at (__n=<optimized out>, this=0x7fffe67fa5e0) at /usr/include/c++/11/bits/stl_vector.h:1094
#10 sick::MessageCreator::createLaserScanMsg (this=this@entry=0x55555568b1a0, data=..., now=<incomplete type>) at /home/ros/tanmay_ws/src/sick_safetyscanners2/src/utils/MessageCreator.cpp:62
#11 0x00005555555823ed in sick::SickSafetyscannersRos2::receiveUDPPaket (this=0x5555555ecb80, data=...) at /home/ros/tanmay_ws/src/sick_safetyscanners2/src/SickSafetyscannersRos2.cpp:88
#12 0x00007ffff7f4bfdc in std::function<void (sick::datastructure::Data const&)>::operator()(sick::datastructure::Data const&) const (__args#0=..., this=0x5555556812c0)
    at /usr/include/c++/11/bits/std_function.h:590
#13 sick::AsyncSickSafetyScanner::processUDPPacket (this=0x55555567e9e0, buffer=...) at /home/ros/tanmay_ws/src/sick_safetyscanners_base/src/SickSafetyscanners.cpp:294
#14 0x00007ffff7f726bc in std::function<void (sick::datastructure::PacketBuffer const&)>::operator()(sick::datastructure::PacketBuffer const&) const (__args#0=..., this=0x55555567ea98)
    at /usr/include/c++/11/bits/std_function.h:590
#15 sick::communication::UDPClient::handleReceive (this=0x55555567ea18, ec=..., bytes_recv=<optimized out>) at /home/ros/tanmay_ws/src/sick_safetyscanners_base/src/communication/UDPClient.cpp:125
#16 0x00007ffff7f72a49 in operator() (bytes_recvd=<optimized out>, ec=..., __closure=0x7fffe67fb360) at /home/ros/tanmay_ws/src/sick_safetyscanners_base/src/communication/UDPClient.cpp:139
#17 boost::asio::detail::binder2<sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>::operator() (
    this=0x7fffe67fb360) at /usr/include/boost/asio/detail/bind_handler.hpp:182
#18 boost::asio::asio_handler_invoke<boost::asio::detail::binder2<sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int> > (function=...) at /usr/include/boost/asio/handler_invoke_hook.hpp:88
#19 boost_asio_handler_invoke_helpers::invoke<boost::asio::detail::binder2<sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>, sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)> > (context=..., function=...)
    at /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:54
#20 boost::asio::detail::handler_work<sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::asio::execution::any_executor<boost::asio::execution::context_as_t<boost::asio::execution_context&>, boost::asio::execution::detail::blocking::never_t<0>, boost::asio::execution::prefer_only<boost::asio::execution::detail::blocking::possibly_t<0> >, boost::asio::execution::prefer_only<boost::asio::execution::detail::outstanding_work::tracked_t<0> >, boost::asio::execution::prefer_only<boost::asio::execution::detail::outstanding_work::untracked_t<0> >, boost::asio::execution::prefer_only<boost::asio::execution::detail::relationship::fork_t<0> >, boost::asio::execution::prefer_only<boost::asio::execution::detail::relationship::continuation_t<0> > >, void>::complete<boost::asio::detail::binder2<sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int> > (handler=..., function=..., 
    this=0x7fffe67fb380) at /usr/include/boost/asio/detail/handler_work.hpp:425
#21 boost::asio::detail::reactive_socket_recvfrom_op<boost::asio::mutable_buffers_1, boost::asio::ip::basic_endpoint<boost::asio::ip::udp>, sick::communication::UDPClient::beginReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::asio::execution::any_executor<boost::asio::execution::context_as_t<boost::asio::execution_context&>, boost::asio::execution::detail::blocking::never_t<0>, boost::asio::execution::prefer_only<boost::asio::execution::detail::blocking::possibly_t<0> >, boost::asio::execution::prefer_only<boost::asio::execution::detail::outstanding_work::tracked_t<0> >, boost::asio::execution::prefer_only<boost::asio::execution::detail::outstanding_work::untracked_t<0> >, boost::asio::execution::prefer_only<boost::as [... cut ...]
  m_intrusion_data_ptr = {<std::__shared_ptr<sick::datastructure::IntrusionData, (__gnu_cxx::_Lock_policy)2>> = {<std::__shared_ptr_access<sick::datastructure::IntrusionData, (__gnu_cxx::_Lock_policy)2, false, false>> = {<No data fields>}, _M_ptr = 0x7fffcc018ed0, _M_refcount = {_M_pi = 0x7fffcc018ec0}}, <No data fields>}, 
  m_application_data_ptr = {<std::__shared_ptr<sick::datastructure::ApplicationData, (__gnu_cxx::_Lock_policy)2>> = {<std::__shared_ptr_access<sick::datastructure::ApplicationData, (__gnu_cxx::_Lock_policy)2, false, false>> = {<No data fields>}, _M_ptr = 0x7fffcc000ca0, _M_refcount = {_M_pi = 0x7fffcc000c90}}, <No data fields>}}

```

puck-fzi, 2024-05-27:
Thanks for the stacktrace! This will hopefully help in fixing the issue.

puck-fzi, 2024-08-21:
Please check if https://github.com/SICKAG/sick_safetyscanners2/commit/c8e4b0d7cdcfbace4e941f52a6991b1f750093a1 resovled the issue

puck-fzi, 2024-09-25:
The fix is included in the newest release 1.0.4, which will be rolled out with the next sync of each distro.

Therefore, this should be fixed, if not please reopen or make a new issue

closed 2024-09-25 by puck-fzi