SICKAG/sick_safetyscanners2 #36: False data on the Sick Nanoscan3 donel

sumitsingh0000, 2024-02-21:
While working with Sick nanoscan3 lidar model, I am getting false data points forming a line. It comes in the range of -1.5rad to -2.32rad always. We have checked the mountings and removed reflective objects nearby. The parameter min_intensities has been kept at 60.

[image]

I am using the parameters:
"frame_id": "front_lidar",
    "sensor_ip": "192.168.15.2",
    "host_ip": "192.168.15.1",
    "interface_ip": "0.0.0.0",
    "host_udp_port": 0,
    "channel": 0,
    "channel_enabled": True,
    "skip": 0,
    "angle_start": -1.98,
    "angle_end": 1.85,
    "time_offset": 0.0,
    "general_system_state": True,
    "derived_settings": True,
    "measurement_data": True,
    "intrusion_data": True,
    "application_io_data": True,
    "use_persistent_config": False,
    "min_intensities": 60.0,

Please suggest the solution for this.

puck-fzi, 2024-08-20:
Any new inputs to this @sumitsingh0000 ?

puck-fzi, 2024-09-25:
Closing since no new input was provided, please reopen if new information arise

closed 2024-09-25 by puck-fzi