SICKAG/sick_safetyscanners2 #40: using nanoScan3 new version ROS2 driver caused lidar scan angle reset to default

SICK-YL, 2024-04-18:
Hello,

I am York from SICK Taiwan,
Recently I got a issue about nanoScan3 new version ROS2 driver caused lidar scan angle reset to default,
Customer used nanoScan3 to achieve some automation application,
They set lidar scan angle 45 to 135 degrees, and used 1.0.3 version ROS2 driver on Linux system,
Lidar scan angle will reset to default value -47.5 to 227.5 degrees,
But used 1.0.2 version ROS2 driver didn't appear this issue,
Maybe there are something we need to notice or check on 1.0.3 ROS2 driver?

Best Regard,
York

puck-fzi, 2024-04-22:
Hi York,

how is the scan angle set, was it set inside the safety designer? 
Then the current version overrrides the data by default. 
In the launch file the following flag needs to be set to use the angles as provided by the safety designer:

` "use_persistent_config": True,`

Hope that helps

puck-fzi, 2024-08-20:
Hi York, did you try it with the updated flag as in the answer?

SICK-YL, 2024-08-21:
Yes, Sir,

Thank for your support, customer already used this flag to fix angle reset issue,
The question I post on Aug.16 is different request from different customer.

puck-fzi, 2024-08-21:
Thanks for letting me know, glad it worked

closed 2024-08-21 by puck-fzi