SICKAG/sick_safetyscanners2 #45: New version NanoScan3 core used 1.0.3 ROS2 driver issue

SICK-YL, 2024-08-16:
Hello Sir,

Recently our customer "FARobot" used 1.0.3 ROS2 driver on NanoScan3 core which FW version was R01.92,
They found the setting angle always reset to default, customer set start angle from -47.5 to 227.5,
But according to lidar diagnostic report the configured angle was -43 to 207,
And customer used two lidar on their AGV, but the strange thing is they can't all use CH1 to transmit raw data,
It needs to set one lidar on CH1 and another lidar set on CH2, otherwise one of them will not transmit raw data,
It was fine if they used previous FW version lidar (R01.65), Is this issue same as Apr.18 I post?
Customer want to know how to fix this angle problem and this issue always happened on 1.92 FW lidar with 1.0.3 ROS2 driver?

Best Regard,
York
[Report_2024-08-12_17-01-28.pdf](https://github.com/user-attachments/files/16631939/Report_2024-08-12_17-01-28.pdf)

SICK-YL, 2024-08-21:
Hi puck,

Customer used new FW device 1.92 and Hardware revision is 1.3.0,
They tried to set angle on launch file and it did not been transfer successfully,
Even they set the angle from -1 to 1, it still failed,
Because they used FW 1.65 device is OK on angle setting,
Maybe the FW upgrade contained the angle setting change or other setting relate to angle?

puck-fzi, 2024-08-21:
Dear York, since the previous firmware version was working with the driver version 1.0.3, I will contact my contacts at SICK to ask whether there were changes in the setup that are not reflected in the driver.

SICK-YL, 2024-08-21:
Hi Puck,

Customer launch file content shown as blew : 
--------------------------------------------------------
sick_safety_f:
    ros__parameters:
      frame_id: "laser_frame_f"
      sensor_ip: "192.168.3.2"
      host_ip: "192.168.3.4"
      host_udp_port: 6060
      channel: 1
      channel_enabled: True
      skip: 0
      angle_start: -0.82
      angle_end: 3.97
      time_offset: 0.0
      general_system_state: True
      derived_settings: True
      measurement_data: True
      intrusion_data: True
      application_io_data: True
      use_persistent_config: True
      min_intensities: 0.0
-----------------------------------------------------
They tried to change " use_persistent_config" into "false", it still can't work,
For your reference.

puck-fzi, 2024-08-21:
Dear York, as in the ROS convention, 0° degrees points towards the front of the sensor; this is in contrast to the SICK scheme. Therefore, the limits provided are out of the allowed area, they customer should sustract 90° (1.57 rad) from the values. However, as long as the flag use_persistent_config: True is set, the values from the safetydesigner should be used. Only if this is set to false, the defined angles of angle_start and angle_end are used.

Could you provide the console output of the launch, there might be an error message stating what is not working, this would help me to identify the issue

SICK-YL, 2024-08-21:
Hi Puck,

Customer feedback there is no error message showed is ok or failed,
They only can check it on "ros topic", they also found the SRC/DST port are the same on FW 1.92 device,
So they need to set CH1 & CH2 and UDP data transmit will ok,
[image]

But  the SRC/DST port are different on FW 1.65 device,
[image]

puck-fzi, 2024-08-21:
Maybe they could try to set the UDP Port in the launch file to 0 to use one of the ports in the ephemeral range to test it. However, I am not sure if this will change anything. I will have a deeper look. Since the firmware is provided directly by SICK, I am not aware of the internal changes in the sensor. For this, I reached out to my contact.

puck-fzi, 2024-08-23:
This is a firmware problem and will be resolved directly via SICK

closed 2024-08-23 by puck-fzi