SICKAG/sick_safetyscanners2 #48: 1.0.3 ROS2 driver angle setting question and Launch file parameter meaning

SICK-YL, 2024-08-27:
Hi Support Team,

According to issue#45, FW1.92 nans3 core used 1.0.3 ROS2 driver may have angle setting issue,
Customer tried to subtract 90° on launch file angle setting and angle set was successful,
But lidar will show M1 error (0x6307000E) about communication interface type problem,
[image]
Customer didn't use the CoLa interface so it was a strange message,

By the way, I have some launch file parameter meaning question,
[image]

1.What is the parameter “channel” means? If we use CH1 to transmit UDP data, It need to set “1” ?
2.Customer tried to set ”2” and it will show interface type error
3.Parameter “channel_enable” need to set ”True” when using UDP data?

Best Regard,
York

puck-fzi, 2024-09-09:
Dear York,

for the first question, I need to clarify with SICK since I am not familiar with the sensor's internal error codes.
The CoLa2 interface will be used by the driver on bootup to transmit data to the sensor and set up the communication.

>  If they used Safety Design angle setting and set "use_persistent_config" to "true", the angle setting will fail,

This should not set new angles and use the originally configured angles of the safetydesigner. So, no new angles should be set. Is that what you experience?

For the launch file questions:
the channels start with 0, the nanoscan has two channels, so channel 0 and channel 1 can be configured. 
The parameter channel enabled needs to be set to true to get data output from the corresponding channel.

Please let me know if that answers the second question

tryit084211, 2024-09-10:
Hi Puck

This is Tom from FARobot, I am working on this issue with our team and cooperating with York
I believe I can show some more information on the situation to better get the situation across

For angle setting, what I am facing is that I can set the angle using ros driver parameter, but not with persistant conf, I tried the following combinations

- case1
  - use_persistant_conf=True
  - start_angle=227.5
  - end_angle=-47.5
  - result: not success with M1 error (angle not supported)
- case2
  - use_persistant_conf=False
  - start_angle=227.5
  - end_angle=-47.5
  - result: not success
- case2
  - use_persistant_conf=False
  - start_angle=3.97
  - end_angle=-0.82
  - result: not success with M1 error (angle not supported)
- case3
  - use_persistant_conf=False
  - start_angle=2.39
  - end_angle=-2.39
  - result: success with M1 error (CoLa)
 
On further investigation, I tried to read the value of start / end angle the driver read from LiDAR, looks like it reads the value we set in SafetyDesigner with no issue (227.5 and  -47.5), although the value itself is not in radient and rotate 90 degree as requested by the driver, but it looks like the driver has taken care of it by using m_angle_offset if I understand it right, so I have no idea what went wrong.

The second issue is actually discovered while investigating the first issue, when I set the the LiDAR using channel 2 in Safety Designer, I can no longer read the settings from LiDAR using the driver, I tried to read and print the LiDAR conf value by using methods defined in ConfigData class, including getStartAngle(), getEndAngle(), getEnabled() and getChannel()
I tried the following combinations

- case1
  - channel_enabled=True
  - channel=1
  - channel in Safety Designer=1
  - result: success (start and end engle as in SafetyDesigner)
- case2
  - channel_enabled=True
  - channel=1
  - channel in Safety Designer=2
  - result: Failed (start and end angle reads out 0)
- case3
  - channel_enabled=True
  - channel=2
  - channel in Safety Designer=2
  - result: Failed ("C2" error (Incompatible configuration))
- case4
  - channel_enabled=True
  - channel=0
  - channel in Safety Designer=2
  - result: Failed (start and end angle reads out 0)
 
[image]

So I just want to know what is the correct way to set this channel (based on your reply it is 0 in driver parameter map to channel 1 in SafetyDesigner and 1 in driver parameter map to SafetyDesigner channel 2) and if the parameter is still working or there is anyhting I did wrong

Thank you in advance for the help

Best,
Tom Liu

puck-fzi, 2024-09-10:
Dear Tom,

For the angle setting, I would have assumed that case 1 and case 4 should be working; in case one, the angles from the safety designer should be used. I will have a look into it. Case 4 is the basic behavior and should work. The other two cases use values outside of the range. The values should be in radians and centered around 0. Since in the ROS world 0 usually points straight ahead.

For the second part, I haven't worked that much with multiple channels; the first case with the 0 channel should be supported. However, I would have expected it to be channel 0 in ROS and channel 1 in the safety designer.

What is confusing me is that I tested the sensor I have just now and was able to configure it with different channels and angles and the persistent configuration. And visualized it in rviz2. There, it seemed to be working.

I first tried in the launch file:
```
"host_udp_port": 0, 
"channel": 0,                                                                      
"channel_enabled": True,                                                                                                                             
 "angle_start": -1.0,                                                               
 "angle_end": 1.0,                                                                  
 "use_persistent_config": False, 
```

That is visualized as a smaller area.

Trial2
```
"host_udp_port": 0, 
"channel": 0,                                                                      
"channel_enabled": True,                                                                                                                             
 "angle_start": -1.0,                                                               
 "angle_end": 1.0,                                                                  
 "use_persistent_config": True, 
```
There, the complete area, as in my safety designer config, was shown.

And when I changed the channel to 1, the same result happened in rviz2, and the output of the log told me it was the second channel, 
When I configured the host_udp_port to a fixed address, I got both channels into the driver, and the visualization was jumping between both channels, so both of them seemed to be working for me.

Could you verify that the behavior is similar for you there?

Additionally, my sensor does not display any M1 error when configuring. 

Could you post maybe the exact launch file you used, so I can try to recreate the behavior? Additionally, which firmware version are you using 1.92?

puck-fzi, 2024-09-10:
Even wireshark shows me the two channels publishing on different ports:
[image]

tryit084211, 2024-09-11:
> And when I changed the channel to 1, the same result happened in rviz2, and the output of the log told me it was the second channel, When I configured the host_udp_port to a fixed address, I got [...]

Actually, I believe there are 3 issues on the table (they might be related to each other)
 - In our setup, we have IPC and two LiDAR wired to the same network switch, when we set both of the LiDAR to use channel 1 in Safety Designer, one of the LiDAR will not send out any UDP data (actually it is not even responding to TCP handshake), we need to set one LiDAR to channel 2 to get data from both LiDAR, this issue happens only in NANS3 with firmware version R01.93 and R01.92 and is already verified as a firmware issue and will be fixed in R01.94
 - After I set one LiDAR to channel to channel 1 and the other to channel 2 (channel in driver both set to 1), I can receive both UDP data from both LiDAR, but the persistant angle config does not correctly applied and it output default full angle range data. And on top of that, driver seems to not able to obtain correct persistant conf from the LiDAR with channel set to 2 in Safety Designer
 - For older LiDAR we have with firmware version R01.65, we got non of the above channel issue, but persistant conf still not taking effect correctly

I can try to recreate the situation you mentioned (one LiDAR send both channel 1 and channel 2 data to the same IPC), I will update it here when I got the result

Our launch file is heavily integrated with our other component, so I might not able to share it, but I can share the parameters we used in the launch file, hope this is enough
```
sick_safety_f:
    ros__parameters:
      frame_id: "laser_frame_f"
      sensor_ip: "192.168.3.2"
      host_ip: "192.168.3.4"
      host_udp_port: 6060
      channel: 1
      channel_enabled: True
      skip: 0
      angle_start: -47.5
      angle_end: 227.5
      time_offset: 0.0
      general_system_state: True
      derived_settings: True
      measurement_data: True
      intrusion_data: True
      application_io_data: True
      use_persistent_config: True
      min_intensities: 0.0

  sick_safety_b:
    ros__parameters:
      frame_id: "laser_frame_b"
      sensor_ip: "192.168.3.3"
      host_ip: "192.168.3.4"
      host_udp_port: 6070
      channel: 1
      channel_enabled: True
      skip: 0
      angle_start: -47.5
      angle_end: 227.5
      time_offset: 0.0
      general_system_state: True
      derived_settings: True
      measurement_data: True
      intrusion_data: True
      application_io_data: True
      use_persistent_config: True
      min_intensities: 0.0
```

puck-fzi, 2024-09-12:
Hi, for the configs, one thing I am noting is that the angle_start and angle_end would need to be in radians to work, not in degrees. However, as long as you use the use_persistent_config = True, this shouldn't matter. However, if you want to set this to false, then you need to set the angles in radians. The range should be [-2.39,2.39]. With 0 pointing straight ahead.

> For older LiDAR we have with firmware version R01.65, we got non of the above channel issue, but persistant conf still not taking effect correctly

This sounds like something in the firmware broke the channels. This is best discussed directly with SICK, I will reach out to them as well. The issue that the persistent config was not working beforehand, I will investigate further.

closed 2025-06-25 by chriseichmann