SICKAG/sick_safetyscanners2 #49: No data being published on topics

ZainCDEI, 2024-12-03:
I am launching the driver with the correct sensor and host IP addresses and it appears to have launched correctly with a few warnings (attached terminals screenshot).

[image]

The topic names also appear in the rqt topic monitor but they don't publish any data. I also tried to call `ros2 topic echo` and `ros2 topic hz` on the **/scan** topic but did not receive any data. 

[image]

I can see the pointcloud being published when I access it through the web interface.

[image]

As part of debugging I played around with completely random IP values in the launch file [here](https://github.com/SICKAG/sick_safetyscanners2/blob/master/launch/sick_safetyscanners2_launch.py#L14-L15) and it printed out the exact same info in the terminal: `Node Configured and running` without any IP related errors or warnings

goodmgk, 2024-12-16:
[image]
[image]

I think I have the same problem.
The topic list is being confirmed, But there is no response when topic echo.
I've been confused by this issue for weeks.

I ask for help my friend

ZainCDEI, 2024-12-16:
I switched to using [this driver](https://github.com/SICKAG/sick_scan_xd) for my multiscan device and also updated the firmware. To update the firmware you need to download the [SOPAS Engineering tool](https://www.sick.com/ag/en/catalog/products/digital-services-and-software/software/sopas-engineering-tool/p/p367244) for windows and upload the firmware file according to your device.

You can first try by switching the driver first and then update the firmware.

goodmgk, 2024-12-16:
[image]

Sorry, the product I'm currently using is sick company nanoscan3.

I'm trying to communicate with ros2 using the sick_safetyscanners2 package.

still..The topic list is being confirmed, But there is no response when topic echo.

how3v3r, 2025-04-22:
Hey I found the REAL problem.
My problem was literally same as you.
real reason was "angle_start" and "angle_end" parameters.
Check when you set them to other values

chriseichmann, 2025-06-25:
Hello,

We are sorry for the late reply. I'm closing this issue as stale. Feel free to reopen it if it's still relevant or open a new one.

Best wishes,
Chris

closed 2025-06-25 by chriseichmann