SICKAG/sick_safetyscanners2 #51: NanoScan3 pro no UDP data output

SICK-YL, 2024-12-23:
Hello Sirs,

Our customer used NanoScan3 pro on AGV application,
And they used ROS2 driver to get lidar UDP data,
Sometimes they found IPC did not get UDP data from nans3,
They tried to power cycle lidar and ROS2 driver, it could solve the problem,
Recently the problem become seriously, lidar did not output UDP data even after power cycle,
And I check the configuration on safety designer, the data output channel was deactivated,
[image]
I also used wireshark to check lidar output data, it seem UDP data did not transmit,
[image]
Data output setting should be configurate when using ROS2 driver to get UDP data?
Customer launch file shown as attachment,
[sick_safetyscanners2_launch.zip](https://github.com/user-attachments/files/18227294/sick_safetyscanners2_launch.zip)

SeungsooLee-Riibotics, 2025-06-05:
Same here, is there any update?

chriseichmann, 2025-06-09:
Hello York, Hello @SeungsooLee-Riibotics,

sorry for the late reply. I would assume you need to set the continuously send mode. I will try to recreate your issue in Hardware on Wednesday. If possible, please provide me with the host (computer) IP address and sensor IP address. The SICK Safety Designer config file and ROS 2 launch file would also help a lot.

Best wishes,
Chris

SeungsooLee-Riibotics, 2025-06-10:
Just to confirm, is "continuously send mode" the same as the "'On request and also continuously to a target computer'" mode in Safety Designer?

Also, for the driver-side configuration, is it enough to set the `host_ip`, `sensor_ip`, and `host_udp_port` parameters in the [launch file](https://github.com/SICKAG/sick_safetyscanners2/blob/master/launch/sick_safetyscanners2_lifecycle_launch.py)? Or are other settings required?

chriseichmann, 2025-06-25:
Here are the files. Please verify that the sensors are ping-able from the host before trying the launch script.

[Safetyscanner2_Issue51.zip](https://github.com/user-attachments/files/20900465/Safetyscanner2_Issue51.zip)

SeungsooLee-Riibotics, 2025-07-07:
Sorry for the late reply.
I've tested with the configuration file you provided, but I was able to reproduce the same issue on our end. The lidar stops sending messages approximately 25 minutes after running **the restarting script**(what I attached above).

What I did and checked.
1. Using safety designer, transferred configuration on my lidars.
2. Update launch file and run once to check launch arguments are working well. (Checked by stdout from node)
3. Run restarting script and ros2 echo to scan.
4. If stuck

I was wondering, is this a common problem that has been reported by others, or does it seem to be a unique case specific to our environment?
Is there anything else you would suggest I try from our side?

SeungsooLee-Riibotics, 2025-07-23:
@chriseichmann
The member device class from nanoscan3-base(https://github.com/SICKAG/sick_safetyscanners) seems like changing data output configuration by itself.
Is there any way to disable overwriting the data output configuration? IMO, if we set the configuration manually through safety designer, overwriting can be redundant and it seems like the reason to make lidar udp stuck. Does it make sense?

antonikaras, 2025-08-15:
Hi everyone,

We're facing a similar issue. We are able to ping the scanner and start the launch file without any errors or warnings. The ```~/field_data``` service contain values on the **field.ranges** area, but we're not getting UDP data. Using Wireshark we can see that there is traffic on the port. From what I can see people had similar issue in the past, was there anyone able to fix it ? Is it on the SICK side or on the ROS2 side ? Should we downgrade the FW version to an older one that it might work ?

ROS2 driver version : 1.0.4 
NanoScan3 pro FW version : 1.93

sensor_ip : 192.168.1.4
udp_port : 6060
host_ip : 192.168.1.1
router_ip : 0.0.0.0